mirkix/BBBMINI #1: Clarification

awright424, 2015-04-19:
Can you provide higher resolution pictures of the front and the back of your cape? how are you powering the beaglebone black? Is it being powered by a esc?

mirkix, 2015-04-20:
The BBB(MINI) is powered by a UBEC 5V / 3A which is connected to the SYS_5V pins (P9.07 and P9.08) [Power supply](https://github.com/mirkix/BBBMINI/blob/master/doc/hardware/hardware.md). High resolutions pictures will follow.

awright424, 2015-04-22:
Is the use of a logic level converter necessary? I have so far a cape with a mpu9250 soldered properly and am waiting for my ms5611. from there I just need to know what else would be necessary. I already have a logic level converter as well.

mirkix, 2015-04-23:
[image]

mirkix, 2015-04-23:
The four ESCs are connected without logic level converter. The GPS is connected with logic level converter, but it can changed to 3.3 Volt I/O also. You should not connect 5 Volt signals to BBB inputs.

awright424, 2015-04-25:
So I got my ms5611 in today and soldered it up. So I now have the mpu9250 and the ms5611. However I can not get ardupilot to detect both correctly. With them both connected Ardupilot automatically fails. With only the mpu9250 it fails on detecting the barometer. running the imu tests in the libraries\AP_IntertialSensor\examples it fails to detect the mpu9250 with both connected and reports a bad crc error for the ms5611. What could be causing this?

mirkix, 2015-04-25:
You have to connected the ms5611 even if you start the AP_IntertialSensor example. Could you post the console output, when you start the example.

awright424, 2015-04-25:
Here is my output when running the test with both modules connected:
ubuntu@arm:~/ardupilot/libraries/AP_InertialSensor/examples/INS_generic$
sudo ./INS_generic.elf
AP_InertialSensor startup...
Failed to boot MPU9250 5 times
Failed to boot MPU9250 5 times
Failed to boot MPU9250 5 times
Failed to boot MPU9250 5 times
Failed to boot MPU9250 5 times
Failed to boot MPU9250 5 times
No INS backends available
Failed to boot MPU9250 5 times

If I disconnect pin 29 on the ms5611 it detects the mpu9250.

starting ardupilot gives me this output with all pins connected minus p9.29
for the ms5611:

./ArduCopter.elf

Init APM:Copter V3.3-dev (5f8c05b7)

Free RAM: 4096
FW Ver: 120
----------------------------------------

load_all took 132us
0 0 0 � Q}�� Q���3�Calibrating barometerz�PANIC:
AP_Baro::read unsuccessful for more than 500ms in AP_Baro::calibrate [2]

root@arm:/home/ubuntu/ardupilot/ArduCopter# ./ArduCopter.elf

Init APM:Copter V3.3-dev (5f8c05b7)

Free RAM: 4096
FW Ver: 120
----------------------------------------

Bad CRC on MS5611
load_all took 128us

the bad crc on MS5611 has me wondering if the chip is the culprit.

mirkix, 2015-04-25:
Does the INS_generic run (the hole test) without problems when the p9.29 is not connected? Do you check the chip select lines (MS5611 P9.42 and MPU-9250 P9.28)? Can you post a picture of your board please.

awright424, 2015-04-25:
With p9.29 disconnected and other pins still connected to the ms5611 does
not interfere with the mpu9250. I can run the INS_generic and calibrate and
test the mpu9250. with p9.29 connected it fails to boot the mpu9250.

AP_InertialSensor startup...
Failed to boot MPU9250 5 times
Failed to boot MPU9250 5 times
Failed to boot MPU9250 5 times
Failed to boot MPU9250 5 times
Failed to boot MPU9250 5 times
Failed to boot MPU9250 5 times
No INS backends available
Failed to boot MPU9250 5 times

mirkix, 2015-04-25:
I just tested the version (5f8c05b7) that you also used, everything works fine on my side. May some wires are wrong. Which kind a breakout boards do you use?

awright424, 2015-04-25:
http://www.ebay.com/itm/251637098006?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT
ms5611

http://www.ebay.com/itm/381029304894?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT
mpu9250

these are the modules i purchased

mirkix, 2015-04-25:
MS5611
Did you connect:
PS to low
SCL to P9.31
CSB to P9.42
SDO to P9.29
SDA to P9.30

awright424, 2015-04-25:
The only thing not connected is ps to low. would that just need to be
connected to ground?

mirkix, 2015-04-25:
Yes, that set the communication mode to SPI, otherwise it is I2C.

awright424, 2015-04-25:
that took care of the issue. ArduCopter says that im ready to fly. I'll
solder up my servo/esc connections and start testing the other portions of
this build

awright424, 2015-04-26:
I've got to bother you with another question. I was wondering If I have this mpu9250 oriented properly? 

[image]

mirkix, 2015-04-27:
I have to look, but you can change it by modifying: https://github.com/diydrones/ardupilot/blob/master/libraries/AP_InertialSensor/AP_InertialSensor_MPU9250.cpp#L288

mirkix, 2015-04-27:
If you use the unmodified SP_InertialSensor_MPU9250.cpp file, the pin 1 dot of the MPU-9250 has to point to the left and to the tail of the copter. 
[image]

awright424, 2015-04-27:
From what you are saying, and with me assuming that vcc is pin 1 then with the way I have things soldered up I would need to either have the ethernet port side of the beaglebone black facing the left of the craft. If necessary I have no problem desoldering the breakouts from the protocape to orient better and possible even lay items out in a more concise manner.
[image]

mirkix, 2015-04-27:
Do you already connect your board to a ground station software via Ethernet? There you can check if the orientation fits. If it not right you could change it in software as described earlier. I add the orientation to the documentation yet.

awright424, 2015-04-27:
No ground control station yet. Couldnt find anything in the gentoo repo's.
Will probably install something on my odroid u3 that is running ubuntu to
use in conjunction with the beaglebone black. Will probably try to get that
going when I get home from work later.

awright424, 2015-05-07:
I switched the orientation of the module after getting it hooked up to mission planner and seeing that it was indeed soldered incorrectly. This is good to go now on my part. I will be getting the power for the beaglebone black and servos situated now.

closed 2015-05-07 by awright424