mirkix/BBBMINI #2: Powering servos and other 5v to 3v items?

awright424, 2015-05-06:
I specifically will be taking my first flight in gas plane and will need to operate at least 4 servos. These are futaba fp-s148 servos. I see that they can operate between 5v and 6v. How would you recommend routing power? I understand you plan on powering the beaglebone black itself through a 5v ubec would you also use this source and just connect the servos to it as well? I know with a quadcopter this is not so much of an issue as you only need to provide ground and signal to the escs.

mirkix, 2015-05-06:
I would recommend to use a second ubec.

awright424, 2015-05-08:
what other hardware will be supported under BBBMINI? Will a buzzer and a way to check voltage be supported?

mirkix, 2015-05-10:
A buzzer could be connected to pin P8.36, but you have to use a power amplifier, because it is a 3.3V low current output. Voltage monitoring is not supported at the moment.

awright424, 2015-05-11:
So are you saying a logic level converter would be necessary for the buzzer?

mirkix, 2015-05-11:
I think a logic level converter will not work here. You have to convert the 3.3V PWM signal into a signal which can driver your buzzer. The necessary part depends on your buzzer.

closed 2015-06-10 by mirkix