mirkix/BBBMINI #3: overlays pru loaded for bbbmini

awright424, 2015-05-11:
I have been looking at the startup.sh located in the bbbmini directory for the device tree's. I have noticed that unlike the pxf or the startup.sh located under linux_hal_essentials that it does not initialize the pru. So at this point I am lost. with the startup.sh under the bbbmini the pru isnt initialized nor is the firmware loaded for them.

mirkix, 2015-05-11:
The PRU will be initialized within this file https://github.com/diydrones/ardupilot/blob/master/Tools/Linux_HAL_Essentials/devicetree/bbbmini/BB-BBBMINI-00A0.dts (https://github.com/diydrones/ardupilot/blob/master/Tools/Linux_HAL_Essentials/devicetree/bbbmini/BB-BBBMINI-00A0.dts). The firmware for PRU is loaded here https://github.com/diydrones/ardupilot/blob/master/libraries/AP_HAL_Linux/RCOutput_AioPRU.cpp#L56.

Run startup.sh to initialize the PRU, the firmware is loaded at runtime.

BBBMINI uses only one PRU for RC input and RC output. So all should work and is already tested.

awright424, 2015-05-11:
The reason I ask is I noticed when running the startup.sh loaded in the root of the ardupilot/Tools/Linux_HAL_Essentials it mentions the fact that the pru is initialized. looking at the dts for the bbbmini I dont see anywhere where the pru firmware is loaded or referenced. I do not see the pru being intialized when loading the bbbmini overlay as well.

awright424, 2015-05-14:
I also have another question about the kernel u=you used during flight. I know a kernel with preempt or rt preempt is needed from my reading. I could not find any .debs or anything for a kernel that the erlerobot project was using. Can you shed some light on the kernel and the dtbo as the rc output isnt working for me. I tried changing servo parameters through mavlink and saw no change.

mirkix, 2015-05-14:
I used this to compile the RT Kernel: http://dev.ardupilot.com/wiki/building-the-code/building-for-beaglebone-black-on-linux/. Did you try the example sketch for RC OUT https://github.com/diydrones/ardupilot/tree/master/libraries/AP_HAL/examples/RCOutput ? This is a good point to start debugging.

mirkix, 2015-05-14:
The first dmesg show that you load the BBBMINI device tree (at 399 s) and then also load the pxf device tree (at 1011 s) you should only load the BBBMINI device tree! 
The 'make: pasm: Command not found' shows that the assembler is not installed on your system. But that is not necessary anyway. The PRU firmware is already build  (https://github.com/diydrones/ardupilot/blob/master/Tools/Linux_HAL_Essentials/pru/aiopru/RcAioPRU_bin.h).

awright424, 2015-05-14:
I have loaded the bbbmini overlay after rebooting and still saw issues in dmesg related to the pru. I am in the process of building the RCoutput example to see if that can shed light on things.

http://pastebin.com/HQ99jb4s  <-- dmesg after loading the overlay.

[  335.303516] omap_hwmod: pruss: failed to hardreset
^ the above line leads me to believe the pru is not being setup correctly. especially after doing some research on the error.

mirkix, 2015-05-14:
this is my dmesg output:

http://pastebin.com/rB3ygHcr

there is also the pruss: failed to hardreset, but that seems to not be a problem.

awright424, 2015-05-14:
hmmm, thats interesting. I tried running the RCOutput.elf not sure if it needed any parameter to be run with or if it needed to be built any differently. Makefile didn't show any options.

mirkix, 2015-05-14:
to build the test sketch you should use `make bbbmini`

the output should look like this:
```
mirko@arm:~/bin$ sudo ./RCOutput.elf 
Starting AP_HAL::RCOutput test
reversing
reversing
reversing
reversing
reversing
^Cmirko@arm:~/bin$ 
```

awright424, 2015-05-14:
I built the aiopru firmware and copied it over to /lib/firmware. I also rebuilt the RCOutput and now get the proper output. I will be home in an hour or two and can then verify that the servos are or are not working. Does this application test all servo outputs?

awright424, 2015-05-14:
I made it home and hooked up my servos to my servo outputs and got nothing. I can hear the servo being powered on but get no movement from the servos.

mirkix, 2015-05-22:
Is the console output like:
```
Starting AP_HAL::RCOutput test
reversing
reversing
```

Maybe the 3.3 Volt signal is to low. I tested the rc output with servos without any problems.

awright424, 2015-05-22:
These are servos I have been able to get working with pwm pins before without using any sort of logic level converter so I dont believe that is the issue. I will more than likely just load the overlay remove the cape and connect signal directly to one of the rc outputs to see if that makes any difference.

mirkix, 2015-05-22:
Could you post the result from `cat /sys/devices/bone_capemgr.9/slots`.

It should look like this:
```
0: 54:PF--- 
 1: 55:PF--- 
 2: 56:PF--- 
 3: 57:PF--- 
 4: ff:P-O-L Bone-LT-eMMC-2G,00A0,Texas Instrument,BB-BONE-EMMC-2G
 5: ff:P-O-- Bone-Black-HDMI,00A0,Texas Instrument,BB-BONELT-HDMI
 6: ff:P-O-- Bone-Black-HDMIN,00A0,Texas Instrument,BB-BONELT-HDMIN
 7: ff:P-O-L Override Board Name,00A0,Override Manuf,BB-BBBMINI
 8: ff:P-O-L Override Board Name,00A0,Override Manuf,am33xx_pwm
 9: ff:P-O-L Override Board Name,00A0,Override Manuf,bone_pwm_P8_36
```

mirkix, 2015-06-03:
There is now a prebuild Debain 8 image with 4.0.4 PREEMPT RT Kernel, GCC 4.9 and device tree already installed. It is not necessary to load any device tree files. Maybe you give it a try. https://github.com/mirkix/BBBMINI/blob/master/doc/software/software.md

awright424, 2015-06-04:
Yes I can test that today when I have a chance. Anyway you would be able to send me one of those spiffy pcb's you added?

awright424, 2015-06-04:
Oh and rcnee may have added your dts file due to me asking questions in #beagle

mirkix, 2015-06-04:
Yes sure, I can send you a Rev 1.0 pcs. Unfortunately there is one pin swap in the layout, but when you know that, it should not be a problem. Just send me your address to mirkix@gmail.com. There will be a Rev 1.1 without the pin swap and cleaner layout soon.

closed 2015-06-10 by mirkix