mirkix/BBBMINI #6: RC_out issues

aaraujo11, 2015-06-13:
Hi @mirkix ,

First of all congratulations for this project, nice work!!!

So i'm trying it with a different platform, i'm using a hexarotor. And so far i successfully got working the IMU MPU9250, Baro MS5611, the GPS Neo-m8 ublox. For the receiver i'm using an FRsky X8R connected to RC_in in the define BBB pins. The FRsky X8R operate with SBUS protocol, it is tested and working. I have also tested the RC_out, and all 6 motors are operating correctly. For last i had configure the APM planner to communicate with the BBB using wireless network.

Basically the issue is, after launch ArduCopter.elf,  when i try to throttle or send other commands in the radio, the motors don't respond to any of them. In the APM planner i can see in the "Radio Calibration" section, the radio commands work fine (roll, throttle, pitch, yaw) as well in the "FailSafe" section in "Radio In" column. But in the "Motor Out" column the corresponding outputs bars remain static. Do you have any clue of the problem is? (or as I have recently start to explore the Ardupilot code, probably could be any configuration i'm missing)

mirkix, 2015-06-13:
Hi @aaraujo11 ,
it sounds a little bit to me, that you did not arm the ArduCopter. Can you check that please. http://copter.ardupilot.com/wiki/flying-arducopter/arming_the_motors/

aaraujo11, 2015-06-17:
Hi @mirkix,

Yeah... "Arm the motors by holding the throttle down, and rudder right for 5 seconds." :laughing: simply a noob at APM.  Thanks. Solved!

closed 2015-06-17 by aaraujo11