mirkix/BBBMINI #8: kernel source, gstreamer with ardupilot.

awright424, 2015-06-22:
I was wondering if there was anyway you would be able to provide the kernel source/headers as a .deb? I need to build wifi drivers for my 80211ac adapter.

I have also recently built gstreamer 1.4.5 on my beaglebone black and have been playing around with streaming. Have you had any experience with this and do you think the beaglebone could handle this and ardupilot? I am using a logitech c920 so the camera does all the grunt of video encoding.

http://docs.emlid.com/Navio-APM/video-streaming/
I was able to stream video to my android phone using the apk provided here. Might be interesting to use this application in conjunction with google cardboard or modified for cardboard for fpv.

Also if you have not mailed the pcb I can pay to have it expedited as I would love to get my hands on it.

mirkix, 2015-06-22:
Hi Alex,

unfortunately I did not create a .deb until now. Maybe you can give me some instructions how to build the .deb, then I will try it. 

The Kernel was build with the `bb-kernel`GitHub project from RobertCNelson. For further instruction look at https://eewiki.net/display/linuxonarm/BeagleBone+Black#BeagleBoneBlack-LinuxKernel .

The video-streaming sound very interesting, but I have not made any video-streaming with a BBB. But keep us informed, it would be a nice feature. 

I mailed the PCB one or two weeks ago, but the German postal goes on strike just here. I also send some PCBs to other people in Germany, they do not arrived there also until now.

Best regards
Mirko

khancyr, 2015-06-25:
Hi ,

If you use bb-kernel , to create .deb, you can use 
``` ./build_deb.sh
```
instead of 
``` ./build_kernel.sh
```

awright424, 2015-06-25:
Mirko,
 Other than hoping the kernel has the ability for me to just zcat /proc/config.gz to get the config was there anything else you did in order to make things work for the pru etc? I dont mind compiling the kernel if necessary. Should be able to build on an x86_64 host with the arm compilers. I know its easier done through a debian based system.

mirkix, 2015-06-28:
Thanks for the information about the `./build_deb.sh`thinks can be so easy. RobertCNelson now also provide a RT-PREEMPT image for the BeagleBone Black. You can install it with `sudo apt-get install linux-image-4.0.6-bone-rt-r5`. With r6 there will be BBBMINI device tree support by the kernel out of the box. That should make thinks much more easier and the BBBMINI image is not necessary any more. The device tree for the BBBMINI is now merged into https://github.com/RobertCNelson/dtb-rebuilder.

awright424, 2015-06-28:
so I installed the version specified and received errors on starting ardupilot: 

RCOutputAioPRU.cpp:SIGBUS error gernerated

mirkix, 2015-06-29:
The device tree is not included in the r5 kernel. So you have to do:
```
git clone https://github.com/RobertCNelson/dtb-rebuilder.git
cd dtb-rebuilder
make clean
make
make install
```
after that you have to add `dtb=am335x-boneblack-bbbmini.dtb` behind `uname_r=4.0.6-bone-rt-r5` in the file `uEnv.txt` than reboot. `uEnv.txt` is located under `/boot`if you are using Debian.

awright424, 2015-06-30:
Thanks that did the trick. I also received the 2 pcb's you sent. ordering another mpu9250 and ms5611 to make sure i didn't damage the boards when desoldering them. I have noticed some device errors, that go away away after reboot.

Thanks again for sending the pcb's I know you said there was something I needed to know when it came to the pcb's themselves. You said some wiring needed to be switched?

mirkix, 2015-06-30:
Yes, right. With PCB Rev 1.0 boards you should NOT use P11 to power the cape and the beagle bone. To power the beagle bone and the cape use +5V pins from P15 or P7 or P12. You can also power the beagle bone and the cape via the power plug of the beagle bone. This is fixed with the Rev 1.2 boards. You should also not close jumper P3.

mirkix, 2015-06-30:
There is now the 4.0.6-bone-rt-r6 kernel available form Robert C Nelson. In this kernel the `am335x-boneblack-bbbmini.dtb`device tree file is already included. You only have to add it to the uEnv.txt file (`dtb=am335x-boneblack-bbbmini.dtb`). `dtb-rebuilder`is not necessary anymore.

mirkix, 2015-07-10:
Hi Alex,

you can use Rev1.0 boards with limitations:

- Do not power the board via P11 use instead +5V pin from P15, P7, P12 to
power the board. Or you can also power the board and bbb via the power plug
of the bbb.
- Please do not close P3
- The Quad-Frame extra RC-Outputs are at the wrong corners

All these things are fixes with Rev1.2.

There is now also a Rev1.4 available (https://github.com/mirkix/BBBMINI-PCB).
I add a level shifter to the RC_IN so you can plug a 5V RX direct to the
bbbmini and also add a Port to connect a HC-SR04 a 2$ (aliexpress) Ultra
Sonic Rangefinder (range 3cm to 400cm, with 1cm accuracy).

Best regards Mirko

mirkix, 2015-07-10:
Sorry answer this twice.

2015-07-10 10:35 GMT+02:00 Mirko Denecke <mirkix@googlemail.com>:

awright424, 2015-07-20:
This can be closed. I was wondering though if you could send me one of the latest pcb's if you have anymore. I am at the point of choosing a receiver and transmitter combo to go with my 9xr pro and would love one of the latest pcb's. If you have paypal let me know I can definitely provide postage plus a tip for your work and help.

closed 2015-07-21 by awright424