mirkix/BBBMINI #9: Software .dtb

awright424, 2015-07-21:
Can you add a section to the software area that deals with building the pru firmware and loading the proper overlay with the 4.0 kernel. I have updated my kernel from your image but seem to be having issues with the overlay loading and the pru's working.

mirkix, 2015-07-26:
Hi Alex,
it is not necessary to build the PRU firmware. If you are using Kernel 4.0 you have to do:
`git clone https://github.com/RobertCNelson/dtb-rebuilder.git` 
`cd dtb-rebuilder`
`git checkout 4.0.x`
`make clean`
`make`
`sudo make install`

After that you have to modify the `/boot/uEnv.txt` file. 

`sudo nano /boot/uEnv.txt`

Add a line:
`dtb=am335x-boneblack-bbbmini.dtb`

Now reboot.

awright424, 2015-07-26:
Thank you will try that out. Is there any chance of me being able to get
the latest revision of the pcb? I will probably try to get something 3d
printed as to not damage the cape and the beaglebone black.

mirkix, 2015-07-26:
I Alex,
the Rev 1.4 boards unfortunately not arrived here until now. I will inform you when they are available.  You can also produce your own batch, take a look on this description:
https://github.com/mirkix/BBBMINI-PCB#pcb

awright424, 2015-08-02:
I finally took a deeper look. the bbbmini is not included the dtb-builder repo it is included the bb-overlays repo. Now I am wondering if I should be cloning that repo and attempting to load the cape versus the dtb. will report back.

mirkix, 2015-08-02:
Hi Alex,

it is in the `dtb-rebuilder` repo as I have written above. https://github.com/RobertCNelson/dtb-rebuilder/blob/4.0.x/src/arm/am335x-boneblack-bbbmini.dts

awright424, 2015-08-02:
it is only available in the 4.0.x branch so cloning and building will not get you the dtb

closed 2015-08-05 by awright424