mirkix/BBBMINI #10: Segment fault

juvinski, 2015-07-28:
Hi,

I'm trying to run Arducopter following the BBB Mini software, but if I try to run any test software or arducopter I'm receiving this segment fault

(gdb) run
Starting program: /home/debian/ardupilot/libraries/AP_InertialSensor/examples/INS_generic/INS_generic.elf 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x000140e2 in Linux::LinuxI2CDriver::LinuxI2CDriver (this=0x33690 <i2cDriver0>, semaphore=<optimized out>, device=<optimized out>)
    at /home/debian/ardupilot/libraries/AP_HAL_Linux/I2CDriver.cpp:36
36	        hal.scheduler->panic("I2C device is not a chardev node");

Any clue ?

Regards

mirkix, 2015-07-29:
Hi,
check if there is a `/dev/i2c-2` device  (`ls /dev/*i2*`).
Best regards
Mirko

juvinski, 2015-07-29:
Hi,
thanks for your reply.
Yes there is two devices on i2c
ls /dev/*i2*
/dev/i2c-0  /dev/i2c-1

I remove the board and wihtout nothing connected still get this error. Btw, i2cdetect comman has output reading i2cbus
i2cdetect -r 0
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-0 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- 34 -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

i2cdetect -r 1
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- 77 

thanks

awright424, 2015-07-29:
Any ideas on this? Or is there a revision that doesnt exhibit this behavior that I can checkout? I was hoping to begin testing and tuning later on today.

juvinski, 2015-07-29:
Hi awright424,

are you with same problem ?

awright424, 2015-07-29:
https://github.com/diydrones/ardupilot/issues/2616

Here is what I have so far and I also have the i2c devices. This issue popped up within the last week or two as I have been thinking it was a dtb issue on my end.

awright424, 2015-07-30:
Any ideas on what is causing this? I have not been able to get off the ground due to this one issue.

juvinski, 2015-07-30:
Hi awright424,

first thing I'm doing is compiling a new kernel and trying debian and ubuntu images to check if the problem is on linux/kernel or ardupilot code, what beaglebone are you using? I'm using beaglebone black revision A6

awright424, 2015-07-30:
juvinski,
 I have a revision C beaglebone black. I dont believe compiling a new kernel will be necessary. I will attempt to use ubuntu myself and then compile the bbbmini overlay/dtb and try to load it at boot. I believe the firmware for the pru needs to be in /lib/firmware I will double check that.

I do not have my serial cable so I will not be able to debug the boot if something goes wrong.

mirkix, 2015-07-30:
Hi,

you need a `/dev/i2c-2` device entry. This is the i2c bus the BBBMINI use.

mirkix, 2015-07-30:
Are you using Debian? And which version / Kernel?

juvinski, 2015-07-30:
Hi Mirko,

I'm using a GY-87 (http://www.goodluckbuy.com/gy-87-10dof-mpu6050-hmc5883l-bmp180-three-axis-gyroscopeacceleratormagnetic-field.html) on i2-c bus.
Debian 8 and 7 sd cards was used and with same problem.
I tryed use Arducopter too(compiling using make linux) and the problem (segment fault) are happening too, this is why I'm suspecting something on linux layer(kernek 3.8.13 bone 70)

mirkix, 2015-07-30:
Hi,

can you try this image please https://github.com/mirkix/BBBMINI/blob/master/doc/software/software.md , that should work.

juvinski, 2015-07-30:
Hi Mario.
That was me first try :(

awright424, 2015-07-30:
That's the image I am using as well.

mirkix, 2015-07-30:
I will check that tomorrow, maybe I have to update the image.

awright424, 2015-07-30:
I have followed your instructions mriko about building the dtb for the
bbbmini and booting with it but that does not work as I have a device that
does not boot using the 4.2 kernel at that point. If you could provide
instructions on building and placing firmware etc for the beaglebone black
then an image would not be necessary but welcomed none the less.

juvinski, 2015-07-31:
Hi,

I was looking the Arducopter code and I think maybe I have a clue.
On my beaglebone I´m with ic2 devices 0 and 1 but Arducopter code for BBBMINI is looking for a i2c device 2 as follow

static LinuxSemaphore  i2cSemaphore0;
static LinuxI2CDriver  i2cDriver0(&i2cSemaphore0, "/dev/i2c-2");

I will try to change this address tomorow and see what will happen.

Regards

awright424, 2015-07-31:
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_HAL_Linux/HAL_Linux_Class.cpp

looking at this file it looks like they were attempting to add both i2c devices? from what i can ascertain I believe it should look like this for the bbbmini.

#elif CONFIG_HAL_BOARD_SUBTYPE == HAL_BOARD_SUBTYPE_LINUX_BBBMINI
static LinuxSemaphore  i2cSemaphore0;
static LinuxI2CDriver  i2cDriver0(&i2cSemaphore0, "/dev/i2c-0");
#else
static LinuxSemaphore  i2cSemaphore1;
static LinuxI2CDriver  i2cDriver0(&i2cSemaphore1, "/dev/i2c-1");
#endif
static LinuxSPIDeviceManager spiDeviceManager;

juvinski, 2015-07-31:
Hi,
what i2c devices you have?
what is the return of ls /dev/i2* command?

awright424, 2015-07-31:
ls /dev/i2c-*
/dev/i2c-0  /dev/i2c-1

juvinski, 2015-07-31:
You can try to change to i2c-1 or 0 compile and try to run?

awright424, 2015-07-31:
yeah i changed it and am compiling to try and run now

awright424, 2015-07-31:
that seems to have fixed the issue. it compiled and started without segfaulting. going to check things out.

juvinski, 2015-07-31:
Nice.

I will do some tests and let you know about the results.

Regards - BTW - I received an email from shen2you and my boards will be shipped tomorow :)

awright424, 2015-07-31:
Are these 1.4 boards I could use one. I need to get a couple. wish it had voltage sensing capabilities as well.

juvinski, 2015-07-31:
What sensors are you using?

earsneyes, 2015-07-31:
Good job guys! I was suffering as well.
Just tested it with a Drotek 10DOF imu (MPU9250, MS5611).

juvinski, 2015-07-31:
Hi,

anyone has problems with ToneAlarm?
Init APM:Copter V3.3-rc8 (818ef282)

Free RAM: 4096
FW Ver: 120
----------------------------------------

load_all took 32us
ToneAlarm: Error!! please check if PWM overlays are loaded correctlyAP_Notify: Failed to initialise ToneAlarmBad CRC on MS5611

mirkix, 2015-07-31:
Thats is ok, bbbmini has no ToneAlarm. Maybe I will remove ToneAlarm for bbbmini, to avoid this message.

juvinski, 2015-07-31:
Hi Mirko,
There someway to avoid that ? My arducopter is not running :(

mirkix, 2015-07-31:
Hi,

just updated https://github.com/mirkix/BBBMINI/blob/master/doc/software/software.md#prepare-bbb . Please check that, it should fix your `/dev/i2c-2` things and enable the Range Finder support.

mirkix, 2015-07-31:
@juvinski 
That is not the reason arducopter is not working. The reason is the `Bad CRC on MS5611`. There is no communication with the MS5611 baro sensor. Do you have a MS5611 connected to the SPI bus?

phoneguy, 2015-08-03:
Hi, 
    I just tested out your latest image and i2c devices now working in Ardupilot.  Thanks!

closed 2015-08-03 by juvinski