mirkix/BBBMINI #12: Receiver question

juvinski, 2015-08-03:
Hi All,

one question - it's not clear to me until now is about the receiver.
What kind of receiver can be used? PPM or PWM will work? On the board there are one RC_Input with Voltage Divider and another 3 inputs with X_Quad name and I couldn't figured out what is the purpose.

Regards

awright424, 2015-08-03:
Need one with ppm sum, cppm or sbus. I use a orangerx r620x connected to bind port

juvinski, 2015-08-03:
Hi, 
This part for me is a little hard.
I have 2 turnigy receivers, 2 dsm2 and one openrls(http://www.hobbyking.com/hobbyking/store/__27096__OrangeRx_Open_LRS_433MHz_9Ch_Receiver.html)
One of this can work?
Thanks

earsneyes, 2015-08-04:
@juvinski PPM or SBUS should work. I successfully tested an Orange 7 Channel DSM2 and a FrySky X8R SBUS receiver. 

The RC_1, RC_2 and RC_4 pins are the outputs needed for a quadcopter. 
Only the RC input will work with the receiver so no PWM input.

Cheers

awright424, 2015-08-04:
If your receiver doesn't do ppm sbus you can get a ppm encoder. But a receiver that does sbus cppm or ppm will be better.

mirkix, 2015-08-04:
@juvinski 
For a connector description look here: https://github.com/mirkix/BBBMINI-PCB#connector .

juvinski, 2015-08-04:
Hi Mirko,
great, now is more clear, except by one thing, P15, P16  and P17 are the motors output for X Quad, but they are for motors, 1, 3 and 4, and how about motor 2 ?

mirkix, 2015-08-04:
[image]

juvinski, 2015-08-04:
Sorry

closed 2015-09-18 by mirkix