mirkix/BBBMINI #14: Quadcopter flipping on take off

awright424, 2015-08-13:
I have gotten my quad together and the beaglebone black is working fine. I however have not been able to fly yet due to it flipping. I am not sure if I have something wired wrong with the motors but everything should be fine. If there is anything you can suggest that  would help. I will probably make a video to show whats happening.

juvinski, 2015-08-13:
Hi,
first of all, lets start from beginning. have you already made the ESC calibration and was ok ?

awright424, 2015-08-13:
I have manually calibrated esc's tried through the beaglebone as well, though I'm not sure it was done right

juvinski, 2015-08-13:
ok, what ESCs are you using ? 
Do you know when can make the video ?

awright424, 2015-08-13:
Emax 12a simonk esc's. Should be able to make one in 3-4 hours. Depending
on when I get home.

earsneyes, 2015-08-13:
there were something reported about Emax 12A esc's incompatibility at least in combination with Pixhawk i think. search diydrones or drone-discuss. 
there is a hardware fix as well (removing a smd component).

awright424, 2015-08-13:
I have looked through the diy drones and only see mention of the 4 in 1 esc nothing concrete stating issues with the 12a simonk esc's. I was also wondering if I need to be using a H quad setup with the emax 250 nighthawk or if I should be using the X quad frame?

juvinski, 2015-08-13:
I have this 12A on a 250 tricopter and one thing for me, I always need to calibrate the esc's at least 2  times to work ok - on first time always one motor or starts after all others or the speed are different.

awright424, 2015-08-13:
i noticed at times all 3 motors start and one twitches a bit before spinning. So I dont know if its loose connections right now or not. I will inspect each connection and see what happens.

El-Mango, 2015-08-13:
What about propellers spinning direction? Did you check?
Just in case...
-e

awright424, 2015-08-13:
I believe I have things set up properly. I will make a video showing how
things are. I am still unsure if I should be running h or x quad setup.

mirkix, 2015-08-13:
Hi Alex,
please also provide the log files which you can find here on your bbb: `/var/APM/logs/`. I had some successful flights today with Stab, Auto and RTL.
Best regards
Mirko

awright424, 2015-08-14:
I woke up and started messing around and realized that I have a revision 1.0 pcb. Are the motor labels correct? Where in ardupilot itself can I see where the motors should be hooked up to what output? I will attach some pictures to this thread.

awright424, 2015-08-14:
[image]
[image]

earsneyes, 2015-08-14:
that should be the problem. The first pcbs had the order wrong. Revision 1.2 and fixed that. 
https://github.com/mirkix/BBBMINI-PCB/blob/master/picture/bbbmini.png
I would try to connect them to the servo connector row on the back side if cable length is enough.

awright424, 2015-08-14:
I have and something still feels off to me. I will be testing the esc's etc
and will have a different flight controller to use in a week or so to test
with as well. I noticed that it seems to want to roll to the right in
stabilize and that pushing forward on the right stick, brain fart, makes it
want to fly backward. So I'm not sure if its the quad or possibly
transmitter doing something retarded

awright424, 2015-08-15:
any ideas guys? I'm at my wits end, only option i can see is ordering a cheap cc3d or something to test with to make sure wiring is good. If I cant wait for this pihat to come I may have to.

juvinski, 2015-08-16:
Hi,
In some cases, you need to invert the pitch channel - if in mission planner you move the stick forward and on mission planner you see the bar for pitch going up, so you need to invert the channel on the radio.
About roll, first - sorry to insist, but have you recalibrate the esc? Have you checked the gravity center ?How is the Balance props ? Vibrations ?
Have you tried trim this roll on transmitter? Was possible to fix ?

awright424, 2015-08-16:
so the issue seems to be the gyro. I got a new one but for some reason if I calibrate for instance and place it level it takes a while for it to actually show as being level in apm planner 2.0

I dont know what else to do. it got off the floor today almost a foot or so before immediately flying towards my front door. I chopped the throttle and looked at apm with it level and it showed it being off in apm. not sure if the barometer could be interfering. I received a replacement mpu9250 and have two more coming. I also have two ms5611's coming as well.

mirkix, 2015-08-16:
Hi Alex,
the best way to solve such a problem is to look into the log files, or provide them to us.

They are located here: `/var/APM/logs`

awright424, 2015-08-16:
https://drive.google.com/file/d/0B4rh1bWqHxtLaXR3dnVTdTkzMWc/view?usp=sharing

here is the latest log file should include the crash as well. not sure why i am getting the behaviour i am from two different gyros. Ive broken two sets of props and am getting frustrated now.

mirkix, 2015-08-16:
Hi Alex,
you can load the log file with APM Planner and look into it. You can see that there are two pulse with throttle input (and output to the esc s). But there is no change in pitch or roll at that time. At the end, when pitch and roll changed, there is no throttle input, maybe because you are lifting the quad up. Maybe it is the wrong log, can you check that?

awright424, 2015-08-16:
https://drive.google.com/file/d/0B4rh1bWqHxtLMl81SjFpZ01tdDA/view?usp=sharing
https://drive.google.com/file/d/0B4rh1bWqHxtLd3BUTHN2eEN5a2s/view?usp=sharing

here are my most recent logs. the latest should show me getting off the ground for a second or two. i smiled. lol

awright424, 2015-08-16:
Oh and in regards to your earlier question I do not remember lifting it if only at the end. Last two logs should have no lifts included

juvinski, 2015-08-17:
Hi Alex,

I'm look your logs and - at least for me, the first and strange thing is you are having something strange with Vibrations - there is just AccZ measurement and is out for regular value - yours are between a and 3.1 instead -15 to -5.
Other strange thing is you just have desired roll and pitch not the current measurement.
How is the value you have in your Log Bitmask ?

awright424, 2015-08-17:
The beaglebone black may not be mounted securely enough to the quad? Or
does it need to be on vibration dampers? What is this bitmask value you
speak of

juvinski, 2015-08-17:
Hi Alex,
this can help about Log Bitmask(http://copter.ardupilot.com/wiki/common-downloading-and-analyzing-data-logs-in-mission-planner/) - I recommend for now enable almost all :)
About Vibrations is extremely hard to analyze but first, there are something strange because you don't have in your logs nothing about roll and pitch and your yaw are with strange values.
Running the tests are everything fine with all sensors ?

awright424, 2015-08-20:
Mirkix,
 I was wondering if the 1.0 board could have anything to do with the issues I have been dealing with? I should be getting the barometers delivered by this weekend as well as some pins etc. Then I would be able to try out a new board to see if the issue was caused by something I'm not seeing.

awright424, 2015-08-20:
So I got my mpu9250 in the mail today. I should have the pins I need coming in to be able to solder up the other 1.0 board I have. Mirkix do you have anymore of the current pcb's? I need to send you some money through paypal if that is the case. I tried out my motors and esc's with a cc3d atom and it flies fine. With that being said I have bought a pixhawk anti-vibration mount as I had the board mounted directly to the frame of the quadcopter. Can you guys show how you have mounted your own beaglebone blacks to a quadcopter frame? I think I may try to design a 3d printed quadcopter frame that integrates mounts for the beaglebone and access holes etc for certain ports.

mirkix, 2015-08-20:
Hi Alex,
you can use Rev 1.0 boards, but be sure you use the right RC_Out to ESC configuration. I saw at your last logs that you fly in acro mode, please make sure to start in stab mode for the first flights. Be sure to mount the breakout boards tight (see picture) to prevent vibration.
[image]
I do not have a Rev 1.4 board at the moment. You can look here, to order a small amount by yourself https://github.com/mirkix/BBBMINI-PCB#pcb.

awright424, 2015-08-20:
how do you have the beaglebone black mounted then? I did not have supports on the outside of the mpu9250 and ms5611 so that may be causing the issues you guys noticed.

mirkix, 2015-08-20:
The BeagleBone Black is hard mounted to the quadframe with nylon spacer, like the mpu9250 and ms5611. Before your next flight, you may disassemble the probs and "listen" to the motors when you arm you copter in stab mode and move the sticks or the copter. The sound must make sense, also take a look to the log or "real" time telemetry in apm-planner.  When all seems ok, mount the probs and try a real flight.

awright424, 2015-08-25:
turns out i have a bad esc which is more than likely the culprit for my issues. I also have been getting unhealthy compass errors. So I have the parts to solder up a new board so I will go ahead and get one going.

awright424, 2015-08-30:
my beaglebone is flying. I no longer have any issues. I have one question. how do you guys start arducopter/plane/etc on boot?

awright424, 2015-08-30:
I also seem to have to constantly calibrate the compass due to it not being healthy. This is very annoying as it means there is no quick way for me to get flying.

earsneyes, 2015-08-30:
I've put a line into my /etc/rc.local to start the rover in a screen session like this:

(sleep 10 && /usr/bin/screen -dmS rover /root/APMrover2.elf -A udp:192.168.1.37:14550 -B /dev/ttyO5 -C /dev/ttyUSB0)

You might need to install screen first: 

`sudo apt-get update && sudo apt-get install screen`.

-A is my gcs udp connection, -B is the GPS module and -C is an 3DR radio connected to the USB-port.

Using two telemetry outputs only works when the ethernet cable is connected as far i can tell.
So im powering up my rover while wired to my home LAN and then head out to my test track, will add a wifi dongle instead of the cable soon.

closed 2015-08-31 by awright424