mirkix/BBBMINI #16: compass needs to be recalibrated all the time

awright424, 2015-08-31:
I am not sure why i have to calibrate my compass every time i try to fly and half the time i cant get it to work properly. Is the resistor on the board for the compass? or is that for the canbus items? I am just not sure why i am forced to constantly re-calibrate the compass when i want to fly. BTW when it is calibrated it flies well. I need to do some tuning and get some fpv going with an android application. But after looking at how far I was able to get it I am wondering how to handle long range wifi.

earsneyes, 2015-08-31:
Not sure why you need to recalibrate each time. You are using a external compass i guess ? 
For the range i will be using a patch antenna and a wifi router.

awright424, 2015-08-31:
examples of antenna and type of router? And no I am not using an external compass the one that is on the board itself.

juvinski, 2015-08-31:
Hi Alex,

It's possible interference?
Have you testes the board off the quad ?

awright424, 2015-08-31:
this is with the beaglebone black being powered through a 5v adapter as well. I am wondering if I need to try to compensate for the motors or anything.

earsneyes, 2015-08-31:
Im using a Fonera router with dd-wrt and a supplied antenna. They are not manufactured/sold anymore i think. 

[image]

[image]

awright424, 2015-08-31:
what would you be using on the quad?

earsneyes, 2015-09-01:
I have a TP-LINK TL-WN727N wifi dongle but can not recommend it because it was little bit of a pain to get it working.

closed 2015-09-10 by awright424