mirkix/BBBMINI #24: Unstable flipping bbbmini

El-Mango, 2015-10-28:
Hi all,

I have a problem with an unstable bbbmini quad. Flipping upside-down both starting from ground or launching by hand.

### The hardware configuration is:
* Frame - Tarot TL300A
* Motor - MultiStar 2206 Baby Beast
* Prop - 5040 Multirotor ABS
* ESC - Afro 12A OPTO V3
* Flight Control - BBB + bbbmini
* Battery - Turnigy nano-tech 1300mAh 3S 45
* UBEC - HKU5 5V/5A
* RX Receiver - X8 R6M 6Ch Micro 2.4GHz Assan
* TX Transmitter - JR BX8103A

### About software:
I followed exactly the steps list on the website (https://github.com/mirkix/BBBMINI/blob/master/doc/software/software.md) 
both, compiling and downloading the 3.3 ArduCopter.elf version.
(from http://firmware.bbbmini.com/ArduCopter.elf)

More or less I've got the same results: No way to fly the copter !

Please notice that I had to modify the parameters, due to my TX:
* RCMAP_PITCH         = 3
* RCMAP_ROLL          = 2
* RCMAP_THROTTLE = 1
* RCMAP_YAW           = 4

The sensations, holding the copter in stabilized mode hanging from my hand, are:

1. Pitch and Roll were exchanged (acting Pitch on the transmitter, made Roll action on drone) even if the representation on the Radio Calibration screen is correct after RCMAP (Pitch with Pitch etc...)

2. Not using transmitter (just mid throttle), stabilizer reaction to manual solicitation was not correct (not equal and contrary to the cause, to keep it stable).

My supposition is that motors are acting in wrong order...
But they are correctly connected !!! No way to go wrong on BBBMINI Printed Board. (They all push up!)

Maybe RC-IN reader piece of code, RCInput_AioPRU.cpp, doesn't handle RCMAP channel substitution ?
Or something else down the line of signal management and motor control?

I'm suspecting of RCMAP because it's the only modification that I made, comparing with standard procedure...
Am I the only one who is trying to use RCMAP channels substitution with bbbmini? (I really cannot change channel number in the TX, only reverse)

What do you thing? Any idea?

Thanks and bye,
Enrico.

earsneyes, 2015-10-28:
Hi Enrico, can you add a picture of your quad maybe ?

El-Mango, 2015-10-28:
Sure I can...
-e
[image]

earsneyes, 2015-10-28:
Thanks Enrico, i think it could be vibration maybe. 
Small copters tend to have strong vibrations so i would look at this as well to see if it is in the acceptable range. 

[http://copter.ardupilot.com/wiki/common-measuring-vibration/](http://copter.ardupilot.com/wiki/common-measuring-vibration/)

I use [Gel pads](http://copter.ardupilot.com/wiki/common-vibration-damping/#gel_pads) and it works pretty well.

mirkix, 2015-10-28:
@El-Mango Hi Enrico, can you check that the upper left motor is connected to RC_2. It seems so, that the motor is connected to RC_1.

earsneyes, 2015-10-28:
@mirkix good catch!

El-Mango, 2015-10-28:
UPS !!!!!
:-|

closed 2015-10-28 by El-Mango