mirkix/BBBMINI #26: Use 4.1 kernel with capemgr

jadonk, 2015-11-19:
Rather than using a replacement .dtb, I recommend using a .dtbo and using the 4.1 kernel. You can use the /opt/scripts/tools/update_kernel.sh script that is on the image (https://github.com/RobertCNelson/boot-scripts/blob/master/tools/update_kernel.sh).

See http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#4.1.x-ti_BeagleBone.2FBeagleBone_Black.2FBeagleBoard-X15_.2B_RT for the flags to pass /opt/scripts/tools/update_kernel.sh.

I used:

    sudo apt-get update
    sudo apt-get install -y lsb-release
    sudo /opt/scripts/tools/update_kernel.sh --ti-rt-channel --testing

I just used a newer console flasher, rather than doing 'sudo apt-get upgrade'. I find using 'sudo apt-get upgrade' removes the usefulness of telling people to grab a particular image because it is unpredictable what might break. Doing the 'update_kernel.sh' script, rather than specifying a version, also adds some unpredictability, but at least I know that is coming from the BeagleBoard.org community, rather than Debian at large.

jadonk, 2015-11-19:
My reference process that I'm documenting as I go: https://gist.github.com/jadonk/ef22de2d04e93022d623

mirkix, 2015-11-19:
@jadonk Hi Jason,
I am using 4.1 already, but before 4.1.13 ti rt r31 released yesterday, CONFIG_SPI_DEBUG=y was enabled and blowing up your log files when ardupilot is running. I am in contact with Robert and there is a DTB file for 4.1 https://github.com/RobertCNelson/dtb-rebuilder/blob/4.1-ti/src/arm/am335x-boneblack-bbbmini.dts available. I will switch to 4.1 on BBBMINI GitHub after the first successful flight.
I think change to `update_kernel.sh` is a good idea.

mirkix, 2015-12-18:
@jadonk 
We are now using `update_kernel.sh`and Kernel 4.1.

closed 2015-12-18 by mirkix