mirkix/BBBMINI #28: Problems compiling new master version

juvinski, 2016-02-14:
Hi,
anyone is trying to compile the new master version?
I'm getting this error 
Waf: Entering directory `/root/ardupilot/build/bbbmini'
[1250/1277] Compiling ArduPlane/GCS_Mavlink.cpp
../../ArduPlane/GCS_Mavlink.cpp: In member function 'void Plane::send_extended_status1(mavlink_channel_t)':
../../ArduPlane/GCS_Mavlink.cpp:154:36: error: 'MAV_SYS_STATUS_REVERSE_MOTOR' was not declared in this scope
         control_sensors_present |= MAV_SYS_STATUS_REVERSE_MOTOR;
                                    ^
compilation terminated due to -Wfatal-errors.

juvinski, 2016-02-14:
An update,
with cross compiling I'm having problems with telemetry.

mirkix, 2016-02-15:
@juvinski Try `waf clean` and `waf`again.

juvinski, 2016-02-15:
I tried but was with error. Removed all folder and cloned again and was ok for now, but I'm having problems with telemetry - is not working in this new version. The parameters - -A and -C has changed?

Regards

mirkix, 2016-02-16:
@juvinski Just checked -A and -C with latest master, works fine.

closed 2016-02-18 by mirkix

juvinski, 2016-02-18:
Hi Mirko,

sure, the problem was on external compass H83XXX.