mirkix/BBBMINI #30: Fresh installation RCOutput issue

aaraujo11, 2016-04-27:
Hi, i have done a fresh installation following the all the steps in software section, but when i'm try to run a simple test (RCOutput.elf) to check the output to the motors, but sometimes the motors start to spin, and other times they simple start to beep more fast, without spin the motors. When i run the main routine (arducopter-hexa) the behavior is the same just beeping. It seems to be a problem in the pwm signal or the pru that isn't active. His there any test to check what is the issue?

In using a hexacopter, with the bbbmini pcb rev.1.4, and DJI 2212 930kv motors and the E300 ESC's.

juvinski, 2016-04-27:
HI Andre,

have you calibrate the radio ?

mirkix, 2016-04-27:
Hi Andre,

do you use Kernel 4.4? `uname -a`

aaraujo11, 2016-04-27:
Hi @mirkix, the kernel version is 
Linux beaglebone 4.4.8-bone-rt-r10 #1 PREEMPT RT Fri Apr 22 04:15:41 UTC 2016 armv7l GNU/Linux

@juvinski do you mean to calibrate the esc's? By set the max and the min os the pwm values?

mirkix, 2016-04-27:
RCOutput.elf is sweeping the PWM value. Some ESCs need a low PWM value for some time because of security reason. RCOutput.elf could be used to test servos but you can get trouble when you use it with ESCs. 
I have to check this Kernel on my system.

aaraujo11, 2016-04-27:
I have it working with the version of December/January. Meanwhile i need to use the BB all this months to other things and i had to flash it. Now that i have some available time, i want to put the hexa flying again.

juvinski, 2016-04-27:
Hi Andre,

Yes, i had problems with some escs beeping until I set the values.

mirkix, 2016-04-27:
Can you try Kernel 4.1 `sudo /opt/scripts/tools/update_kernel.sh --bone-rt-kernel --lts-4_1` and reboot if this helps.

aaraujo11, 2016-04-27:
@mirkix apparently after downgrade the kernel to the 4.1 the RCOutput test work fine, but as @juvinski refer i have to calibrate the radio to define the min and max values, to stop beeping after launch the main routine (arducopter-hexa). I will done some tests before close this issue. Anyway thanks guys for the support! Cheers!

mirkix, 2016-04-28:
@aaraujo11 Checked with Kernel 4.4 without issues here.

aaraujo11, 2016-04-28:
Hi @mirkix, yes i upgrade again to the kernel 4.4, and it is working! Probably as sometimes it works and other times doesn't, i wrongly assume that it has a problem with the kernel, but never thought it will a problem of radio calibration, once the RCOutput test routine supposedly sweeps all ranges of values. But the realty is that with kernel 4.1 RCOutput test routine works, without doing any radio calibration. Anyway thanks for the support!

Just a few questions:

1) For the acc calibration, once in the APM planner, it always give a timeout, a possible way to set the offsets, is run the INS_generic.elf, and calibrate it and get the offset values and insert it directly in APM planner params list, is it the correct way to do it?

2)  What do you use for communicate with or BB? Have your create a wifi access point? Or do use a serial bluetooth module? I have read some comments that using wifi isn't ideal because it may interfere with radio receiver, since they operate in the same frequency 2.4GHz (if you aren't use the new 5GHz), have you experience any interference?

3) I notice in one of your youtube videos you have a status display, in your quad, is there the code available in ardupilot community, or have you develop it?

juvinski, 2016-04-28:
Hi Andre,
I think some questions I can answer.
The Oled display is available at ardupilot main respository and uses a I2C oled  display - it's cool and enabled by default.
About connection, I'm always using 3dr compatible telemetry module - frequency is 915 Mhz as main interface connection. I tried wifi with ad hoc support and works fine too. Regarding interference, is always a risk.

aaraujo11, 2016-04-28:
Hi Vinicius, it is really useful the display info the get the status, i have one oled module of that i will try it! Regarding the connection once i don't have a telemetry module, i will try to config a wifi AP. Thanks!

closed 2016-05-02 by aaraujo11