mirkix/BBBMINI #36: Mission planner doesn't display full parameter list

nikola-v, 2017-01-14:
Hi all, 

installed ArduPlane-3.7.1 and Mission Planner 1.3.43. After syncing through Mavlink there is no Full Parameter list page in Mission Planner. Whole installation and setup was done from the tutorial (https://github.com/mirkix/BBBMINI/blob/master/doc/software/software.md).

mirkix, 2017-01-17:
@nikola-v what kind of connection do you use?

nikola-v, 2017-01-17:
@mirkix  I have tried both UDP and Serial. 
There is no option like in the picture (random google searched pic)
 http://ardupilot.org/copter/_images/RangeFinder_LIDARLite_MPSetup.png
Should I have that option ?

mirkix, 2017-01-19:
http://diydrones.com/forum/topics/mission-planner-1-3-42-no-full-parameters-list

nikola-v, 2017-01-21:
Thank you mirkix!

closed 2017-01-21 by nikola-v