mirkix/BBBMINI #37: Xbee to Mission Planner

nikola-v, 2017-01-15:
Hi all,

I am trying to connect Xbee Pro S3B to Mission Planner. What I have achieved is to connect and verify that it all works through Minicom at baudrate of 57600. I can send data in both ways and all works fine. If I try to connect arduplane to Mission Planner it doesn't work.
Command used is: ./arduplane -C /dev/ttyO4 and that same command works with Prolific USB ttl serial module without any issue. Is there some modification in the code that needs to be done to make Xbee work.

juvinski, 2017-01-16:
Hi nikola have you tried use -A?

mirkix, 2017-01-17:
@nikola-v is it working via udp?

nikola-v, 2017-01-17:
I have reinstalled MP and found out that there is Serial0-5 and I have set all to 57600 (was from 38400 to 115200). It seems that now it works.
@juvinski  I did it with -A and with -C
@mirkix yes it was working with UDP without any issue.

Thank you guys for fast response.

closed 2017-01-17 by nikola-v