mirkix/BBBMINI #38: dtb issue? not sure

awright424, 2017-03-22:
Used the latest beaglebone black image and grabbing arducopter and compiling on debian leaves me with this error when attempting to start ardupilot. I have verified as well that the dtb is being loaded initially.
`[ 2104.096701] Unhandled fault: external abort on non-linefetch (0x1818) at 0xb6fe2000
[ 2104.096707] pgd = dc308000
[ 2104.096717] [b6fe2000] *pgd=9c354831, *pte=4a324343, *ppte=4a324833
[ 2112.307359] Unhandled fault: external abort on non-linefetch (0x1818) at 0xb6fd6000
[ 2112.307365] pgd = dcd38000
[ 2112.307376] [b6fd6000] *pgd=9c451831, *pte=4a324343, *ppte=4a324833
[ 2116.818783] Unhandled fault: external abort on non-linefetch (0x1818) at 0xb6fe8000
[ 2116.818789] pgd = dcf40000
[ 2116.818799] [b6fe8000] *pgd=9cd64831, *pte=4a324343, *ppte=4a324833
[ 2530.019735] Unhandled fault: external abort on non-linefetch (0x1818) at 0xb6f80000
[ 2530.019740] pgd = ddc64000
[ 2530.019751] [b6f80000] *pgd=9c406831, *pte=4a324343, *ppte=4a324833`
 RCOutputAioPRU.cpp:SIGBUS error gernerated

awright424, 2017-03-23:
using the image mentioned in software works. so it appears the pru firmware is not available or needs to be built and installed on the newest image i would assume.

awright424, 2017-04-06:
Can you provide instructions on where the pru firmware needs to be etc? Is it just compiling said pru firmware and copying to /lib/firmware? I would like to use newer images and not be forced to have to use the image specified in the instructions.

juvinski, 2017-04-06:
Hi, 

The dtb are been charged at boot time ?

awright424, 2017-04-06:
If I use the iot debian image and install realtime kernel etc its fine. but it doesnt appear that the pru firmware is loaded. I have not confirmed the  bbbmini dtb is loaded but it has been added using the sed command you provided. I figured I would be able to build the pru firmware and install it myself.

mirkix, 2017-04-15:
@awright424 BeagleBone Blue is running the IoT image and the firmware in running fine.

awright424, 2017-04-15:
Is the beaglebone black still using a bbbmini specific dtb and pru as I
believe I saw a blue specific dtb and pru. Even so for those who would like
to use other distros, arch Linux etc it would be great to have better
instructions as to where the pru firmware built would need to reside.

awright424, 2017-04-22:
`[alex@archlinux ~]$ ssh debian@192.168.0.8
Debian GNU/Linux 8

BeagleBoard.org Debian Image 2017-03-19

Support/FAQ: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian

default username:password is [debian:temppwd]

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Apr 22 15:22:22 2017 from 192.168.0.5
debian@beaglebone:~$ ls
arducopter-quad  bin
debian@beaglebone:~$ sudo ./arducopter-quad 
[sudo] password for debian: 
RCOutputAioPRU.cpp:SIGBUS error gernerated

debian@beaglebone:~$ 
`

this doesnt appear to be the case. I have even mentioned the need for instructions for manually compiling and installing the pru firmware. I know it should be more than just enabling the bbbmini dtb. this is the latest bone-debian-8.7-iot-armhf-2017-03-19-4gb.img

This really causing me alot of headaches as I would like to use other distributions on beaglebone black with bbbmini.

mirkix, 2017-04-23:
@awright424 try to remove / delete `/etc/modprobe.d/pruss-blacklist.conf`

closed 2017-05-03 by mirkix

jackcarrozzo, 2018-07-10:
I'm having this problem too, followed all the directions as best I can and still can't figure out what I missed. I don't have an /etc/modprobe.d/pruss-blacklist.conf

How can I confirm if my PRU DTB is loading?

awright424, 2018-07-10:
Check your uEnv.txt file located in boot. That loads the correct dtb that
was my issue with pocketpilot a day ago.

jackcarrozzo, 2018-07-10:
Per instructions I have in my uEnv.txt:

`uname_r=4.9.105-ti-rt-r113
dtb=am335x-boneblue.dtb
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo`

I can confirm also that /lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo exists. 

Appreciate any other leads you can give me!

awright424, 2018-07-10:
uname_r=4.9.88-ti-rt-r111
enable_uboot_overlays=1
uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
enable_uboot_cape_universal=1
cmdline=coherent_pool=1M net.ifnames=0 quiet

your uEnv.txt needs to be something like this. You have the wrong dtbo being loaded.

jackcarrozzo, 2018-07-10:
I get the same error using /lib/firmware/AM335X-PRU-UIO-00A0.dtbo, unfortunately.