mirkix/BBBMINI #41: SPI unable to open

OscarJGV26, 2017-09-03:
Hi, 

I am having the following error...

root@beaglebone:/home/debian# /home/debian/arducopter -C udp:192.168.137.1:14550
SPI: unable to open SPI bus /dev/spidev2.1: No such file or directory

I checked the /dev folder and spidev2.1 is not there,
[image]

I would assume is just a matter of changing it but I am thinking maybe I missed something. I ran all the steps in (https://github.com/mirkix/BBBMINI/blob/master/doc/software/software.md). 

Any suggestions?

Regards,
Oscar

patrickpoirier51, 2017-09-04:
Looks like you do not have the correct binary... this one is looking for SPI connected sensors that are on the BBBMINI not on the BBBlue
You should use gitter for these type of question: https://gitter.im/mirkix/BBBMINI
Regards

OscarJGV26, 2017-09-04:
Hi patrick, 

Yes. This was my bad. I followed the suggestions in ([1]https://github.com/mirkix/BBBMINI/blob/master/doc/software/software.md) to the letter but they are for BBBMINI as you suggested. 

I corrected this by avoiding step 16 in [1] and selecting board=blue in step [4] of compiling.

I got confused because this process is not described in ([2] https://github.com/mirkix/ardupilotblue) since it provides the image but I wanted to build it for my self. 

As for the No INS backend, I don't know what was exactly but when I ran all the steps in [2] then all the compiling steps in [1], it worked.

Regards,
Oscar

juvinski, 2017-09-04:
hi @OscarJGV26 ,

BBBMini use spi for 9DOF sensor while BBBBlue use just i2c for both 9DOF and Baro, that is the reason you are having this troubles.

OscarJGV26, 2017-09-04:
hi Juvinski,

Yes. I realized I made that mistake on selecting board=blue and also, I assume, loading wrong cape layout in step 16. 

Thank you very much.

closed 2018-01-04 by mirkix