mirkix/BBBMINI #45: libm.so.6: version `GLIBC_2.27' not found

HomeLH, 2019-07-21:
`./ardusub: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by ./ardusub)
`
Crosscompile master computer runs on Ubuntu 18.04.

mirkix, 2019-07-21:
@HomeLH Hi, yes that is an issue because Ubuntu 18.04 is using an newer GLIBC. You can download http://releases.linaro.org/components/toolchain/binaries/4.9-2017.01/arm-linux-gnueabihf/gcc-linaro-4.9.4-2017.01-i686_arm-linux-gnueabihf.tar.xz and put the bin folder in your PATH before the Ubuntu arm compiler. With taht you should be able to compile ArduPilot with Ubuntnu 18.04 for BBBmini. I will check that and but in in the doc.

HomeLH, 2019-07-27:
@mirkix Thanks very much. It does work well, the detailed log:
`hogan@hogan-ThinkPad-T480:~/ardupilot$ ./waf configure --board=bbbmini
Setting top to                           : /home/hogan/ardupilot 
Setting out to                           : /home/hogan/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : bbbmini 
Checking for program 'arm-linux-gnueabihf-ar' : /opt/gcc-linaro-4.9.4-2017.01-i686_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-ar 
Using toolchain                               : arm-linux-gnueabihf 
Checking for 'g++' (C++ compiler)             : /opt/gcc-linaro-4.9.4-2017.01-i686_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ 
Checking for 'gcc' (C compiler)               : /opt/gcc-linaro-4.9.4-2017.01-i686_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc 
Checking for program 'arm-linux-gnueabihf-pkg-config' : /usr/bin/arm-linux-gnueabihf-pkg-config 
Checking for need to link with librt                  : not necessary 
Checking for 'lttng-ust'                              : not found 
Testing link with lttng-ust                           : yes 
Checking for library dl                               : yes 
Checking for 'libiio'                                 : not found 
Testing link with libiio                              : yes 
Checking for HAVE_CMATH_ISFINITE                      : yes 
Checking for HAVE_CMATH_ISINF                         : yes 
Checking for HAVE_CMATH_ISNAN                         : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE        : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE           : no 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE           : no 
Checking for header endian.h                          : yes 
Checking for header byteswap.h                        : yes 
Checking for program 'python'                         : /usr/bin/python 
Checking for python version >= 2.7.0                  : 2.7.15 
Checking for program 'python'                         : /usr/bin/python 
Checking for python version >= 2.7.0                  : 2.7.15 
Source is git repository                              : yes 
Update submodules                                     : yes 
Checking for program 'git'                            : /usr/bin/git 
Checking for program 'arm-linux-gnueabihf-size'       : /opt/gcc-linaro-4.9.4-2017.01-i686_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-size 
Benchmarks                                            : disabled 
Unit tests                                            : enabled 
'configure' finished successfully (1.014s)
...
[1151/1160] Linking build/bbbmini/lib/libAntennaTracker_libs.a
[1152/1160] Linking build/bbbmini/lib/libArduCopter_libs.a
[1153/1160] Linking build/bbbmini/lib/libArduPlane_libs.a
[1154/1160] Linking build/bbbmini/bin/ardurover
[1155/1160] Linking build/bbbmini/bin/antennatracker
[1156/1160] Linking build/bbbmini/bin/arducopter-heli
[1157/1160] Linking build/bbbmini/bin/arducopter
[1158/1160] Linking build/bbbmini/bin/arduplane
[1159/1160] Linking build/bbbmini/lib/libArduSub_libs.a
[1160/1160] Linking build/bbbmini/bin/ardusub
Waf: Leaving directory `/home/hogan/ardupilot/build/bbbmini'

BUILD SUMMARY
Build directory: /home/hogan/ardupilot/build/bbbmini
Target               Text    Data  BSS    Total 
------------------------------------------------
bin/ardurover        653173  1176  30024  684373
bin/antennatracker   516238  1128  28772  546138
bin/arducopter       848401  1200  35680  885281
bin/arducopter-heli  823576  1200  35536  860312
bin/arduplane        858979  1196  36596  896771
bin/ardusub          707163  1192  31496  739851

Build commands will be stored in build/bbbmini/compile_commands.json
'build' finished successfully (1m43.532s)
Waf: Entering directory `/home/hogan/ardupilot/build/bbbmini'
Waf: Leaving directory `/home/hogan/ardupilot/build/bbbmini'

BUILD SUMMARY
Build directory: /home/hogan/ardupilot/build/bbbmini
Target       Text    Data  BSS    Total 
----------------------------------------
bin/ardusub  707163  1192  31496  739851
bin/ardusub  707163  1192  31496  739851

Build commands will be stored in build/bbbmini/compile_commands.json
'sub' finished successfully (1.325s)
`

Yatin-Khurana, 2022-09-25:
can you please post binaries here?